revtel/react-native-nfc-manager #487: Issue NdefFormatableHandlerAndroid - format Ndef - 3.12.0

xgrimaldi, 2022-01-20:
Hi @whitedogg13,

I've tried new update 3.12.0 for NdefFormatable functionality and I found 2 issues.

**1st Issue:**

On requestTechnology function with NdefFormatable tech, as soon as the tag is recognized, app is crashing. 
I could'nt get any error log, since the try catch, isn't catching anything and console.log doesn't show anything after the call on this function. I've seen also a weird thing, when i try to print NfcTech.NdefFormatable in console.log, it show nothing (not like the others). 😮 

*How to replicate :*
- Use requestTechnology function with NfcTech.NdefFormatable
```javascript 
try {
  console.debug("START LISTENING");
  await NfcManager.requestTechnology([NfcTech.NdefFormatable]);
  console.debug("LISTENING TAG : SUCCESS");
} catch (ex) {
  console.debug("LISTENING TAG : ERROR");
  console.warn(ex);
} finally {
  NfcManager.cancelTechnologyRequest();
}
```
- Scan a tag, on Android smartphone NFC reader
- App is crashing directly, without any log.

**2nd Issue :**
Since it's not working with NdefFormatable tech, i tried to read with other tech like Ndef. I was able to read my "NdefFormatable" tag.
I tried to write it with NdefFormatable Handler but it's not working i have the following issue in log : 
```
TypeError: undefined is not an object (evaluating '_reactNativeNfcManager.default.ndefFormatableHandlerAndroid.formatNdef')
```

Maybe it is a 'normal' issue since i'm not using NdefFormatable tech, but i'm not sure, since in old version, I didn't need to specify the technology used and i was working.

*How to replicate :*
```javascript
try {
  await NfcManager.requestTechnology([NfcTech.Ndef]);
  const bytes = Ndef.encodeMessage([Ndef.textRecord('hello nfc')]);
  await NfcManager.ndefFormatableHandlerAndroid.formatNdef(bytes, {readOnly: true});
} catch (ex) {
  console.warn(ex);
} finally {
  NfcManager.cancelTechnologyRequest();
}
```

I'll try on a new project app, without any other package/dependencies, to be sure i've the same behavior.
Can you help to solve this issue ? Thanks a lot. 😸

whitedogg13, 2022-01-20:
@xgrimaldi It looks like the library isn't successfully upgraded to new version in your project.

Please check your local `node_modules/react-native-nfc-manager` directory. For example, you can check for the version in `package.json`, or `src/NfcTech/NdefFormatableHandlerAndroid.js` (old version doesn't contain this file)

FYI, I have also included this feature into our demo app, you can take a look [here](https://github.com/revtel/react-native-nfc-rewriter/commit/caef704927f9de1806b8b74db51a099744ff45b0)

xgrimaldi, 2022-01-20:
I'm pretty sure it was well upgraded. Package was in 3.12.0 and i was able to check the definition of the function. 
I'll check it and i'll try on the demo app also. :wink:

Thanks for quick answer.

xgrimaldi, 2022-01-24:
Hi @whitedogg13 ,
You were right. It seems to be an error on my side. I've tried on the Demo App by clone your project, and it's working well. 👌 
I think you should upgrade the apk on the Google Play Store with this new features, it could be useful for many users. 
Many thanks, i close this (false) issue.

closed 2022-01-24 by xgrimaldi