revtel/react-native-nfc-manager #492: Ios Read tag is not working when app is in kill state

Krish143git, 2022-02-07:
Hi i implemented nfc feature write and read it is working.

But In Ios ,The device is not able to read the tag (There is one  url in the tag) when app is in quit state.It is working in android.Any one please help me how to proceed

venkatakrishna-kiwi, 2022-02-10:
working

whitedogg13, 2022-02-10:
Hi @Krish143git if you're using `NDEF` with `RTD_URI`, you can simply register a `custom url scheme` along with React Native's built-in [Linking](https://reactnative.dev/docs/linking) module, and it will work properly. However, in this case, you won't be able to retrieve the full content for your NFC tag, only the parsed custom URL will be sent to your app.

If you need to access the whole tag content (which is called `background tag reading` for iOS), you will need to first perform some set-up for [universal link](https://developer.apple.com/library/archive/documentation/General/Conceptual/AppSearch/UniversalLinks.html), 

then you should use `NfcManager.getBackgroundTag` for the initial tag scanned:

```javascript
          // get the initial launching tag
          const bgTag = await NfcManager.getBackgroundTag();
          if (bgTag) {
            onBackgroundTag(bgTag);
          }
```

And this code snippet can be used for the following tags:

```javascript
          // listen to other background tags after the app launched
          NfcManager.setEventListener(
            NfcEvents.DiscoverBackgroundTag,
            onBackgroundTag,
          );
```

Krish143git, 2022-02-15:
@whitedogg13 Thank you so much

closed 2022-02-15 by Krish143git