revtel/react-native-nfc-manager #494: Disable apple pay

freeridre, 2022-02-15:
Hello devs,
Are there any way to disable apple pay, while I use my ndef app? Because if my Iphone is close to the nfc reader my apple pay gui opens, and I do not want that...

closed 2022-06-05 by github-actions[bot]

mohmmadmoussa1988, 2022-11-15:
Any update, is there any solution for the stated problem?

bmweapplinse, 2022-12-16:
@freeridre Did you find any work around for this?

mohmmadmoussa1988, 2022-12-16:
@bmweapplinse yes,
you should send an email to 
apple-pay-inquiries@apple.com
apple-pay-provisioning@apple.com

having that info at the header

Company: xxxx
Team ID: xxx
Apple ID: xxxx
Application ID: xxxx

then explain your need, I did this and they provided me with needed entitlement

bmweapplinse, 2022-12-16:
> @bmweapplinse yes, you should send an email to [apple-pay-inquiries@apple.com](mailto:apple-pay-inquiries@apple.com) [apple-pay-provisioning@apple.com](mailto:apple-pay-provisioning@apple.com)
> 
> having that info at the header
> 
> Company: xxxx Team ID: xxx Apple ID: xxxx Application ID: xxxx
> 
> then explain your need, I did this and they provided me with needed entitlement

And then how to use that entitlement from the app side in react native?

mohmmadmoussa1988, 2022-12-21:
@bmweapplinse  still didnt find a way to do make it with reactnative, any update from your side?

BhavyaKoshiya, 2023-08-03:
@mohmmadmoussa1988 @freeridre Did you manage to add this in the app?