revtel/react-native-nfc-manager #501: TypeError: null is not an object (evaluating 'NativeNf Manager[name]')

martin12345678900, 2022-03-11:
Hello, i have the following problem. When i open my react-native expo app from Expo Go using my android device on NfcManager.start() im getting the following error: TypeError: null is not an object (evaluating 'NativeNf Manager[name]'). Please someone help me to solve this issue <3

ahirwar-ankit, 2022-03-19:
+1
I'm Also facing the same problem. Cannot start `NfcManager()`.

whitedogg13, 2022-03-20:
Hi @martin12345678900 & @ahirwar-ankit while using `Expo Go`, you **CANNOT** add `custom native modules` like this one. 

You should consider to use `Expo EAS` if NFC is required for you app, and the related doc is [here](https://github.com/revtel/react-native-nfc-manager#expo)

closed 2022-03-20 by whitedogg13