revtel/react-native-nfc-manager #507: Unable to Read ISO 15693 RF interface (iOS)

TheKeenestKeen, 2022-04-07:
I am trying to read from the following NFC tag with description: "Dynamic NFC/RFID tag IC with 64-Kbit EEPROM, energy harvesting, I2C bus and ISO 15693 RF interface"

As such, I expect that I need to use NfcTech.Iso15693IOS as a value.  

```javascript
async function readNdef() {
        try {
          // register for the NFC tag with NDEF in it
          await NfcManager.requestTechnology(NfcTech.Iso15693IOS);
          // the resolved tag object will contain `ndefMessage` property
          const tag = await NfcManager.getTag();
          console.warn('Tag found', tag?.type);
        } catch (ex) {
          console.warn('Oops!', ex);
        } finally {
          // stop the nfc scanning
          NfcManager.cancelTechnologyRequest();
        }
      }
```

I am unable to get any read on the NFC device. However if I use NfcTech.Ndef, it recognises something, but an error is immediately thrown. Am I missing something here? 

Any help would be greatly appreciated.

closed 2022-04-08 by TheKeenestKeen

gabrielew, 2022-04-18:
did you get some results?
@TheKeenestKeen

TheKeenestKeen, 2022-04-19:
Hi @gabrielew Our hardware engineer sent me the incorrect datasheet (the tag I was reading was a NFC Forum Type 4 tag), so I closed this issue.