revtel/react-native-nfc-manager #508: Issue with NfcManager.requestTechnology(NfcTech.Ndef) and Expo

TheKeenestKeen, 2022-04-10:
Hi @whitedogg13. I purchased your course yesterday and I am able to follow everything step for step until I reach the following line
```javascript
await NfcManager.requestTechnology(NfcTech.Ndef)
```
which throws this error:

`at http://192.168.88.35:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false&minify=false:545956:321 in _createSuperInternal
at http://192.168.88.35:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false&minify=false:545967:25 in NfcErrorBase
at http://192.168.88.35:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false&minify=false:545956:321 in _createSuperInternal
at http://192.168.88.35:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false&minify=false:545997:26 in SecurityViolation
at node_modules/react-native-nfc-manager/src/NfcError.js:103:15 in buildNfcExceptionIOS
at http://192.168.88.35:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false&minify=false:546461:38 in handleNativeException$`

I am however able to get a read using
```javascript
NfcManager.setEventListener(NfcEvents.DiscoverTag, (tag: any) => {
    ...
})
```
However, the approach above only allows for reading and not writing. What is causing this issue?

For context, I am using Expo with the Development Client build with react-native-nfc-manager as a plugin.

whitedogg13, 2022-04-11:
Hi @TheKeenestKeen thanks for supporting!

The error you mentioned is an exception from iOS: `tagResponseError` (which means the actuall error happens in the NFC tag itself, but reported by iOS), and I think using Expo should be just fine.

Please check:
- what tag do you use?
- has the tag been formatted to NDEF yet?

BTW, the course has its own discord channel `nfcs-with-react-native` (in newline's discord server), or you're also welcome to send me DM via Twitter, mine handle is `@richie13tw`, I will be both places to help!

TheKeenestKeen, 2022-04-11:
Hi @whitedogg13 glad to support! 

I can write into the tag successfully using your NFC App on the Appstore. For instance, this is text payload I just wrote: "hello there world".

This is the response object I got  using event listener approach from my App: 
`
Tag found, Object {
  "ndefMessage": Array [
    Object {
      "id": Array [],
      "payload": Array [
        2,
        101,
        110,
        104,
        101,
        108,
        108,
        111,
        32,
        116,
        104,
        101,
        114,
        101,
        32,
        119,
        111,
        114,
        108,
        100,
      ],
      "tnf": 1,
      "type": Array [
        84,
      ],
    },
  ],
}`

However, even with this correctly formatted NDEF, the
```javascript
await NfcManager.requestTechnology(NfcTech.Ndef)
```
approach still shows the exception detailed in my first post.

As for the tag I am using, here is some info:

ST25TA512B,  ST25TA02KB ST25TA02KB‑D,  ST25TA02KB‑P Datasheet

- NFC Forum Type 4 Tag IC with up to 2-Kbit EEPROM
- ISO/IEC 14443 Type A
- Support NDEF data structure

whitedogg13, 2022-04-12:
Hi @TheKeenestKeen you're using the NFC `type 4` tags. 

NFC `type 4` (`IsoDep`) tags have different behavior from `tag 2` (`NfcA`) tags, which are used in our course as demo.

In order to enable NDEF support for `type 4` tags, you will have extra setup to do. You will need to update `Info.plist` with `com.apple.developer.nfc.readersession.iso7816.select-identifiers`:

```
        <key>com.apple.developer.nfc.readersession.iso7816.select-identifiers</key>
	<array>
		<string>D2760000850100</string>
		<string>D2760000850101</string>
	</array>
```

The full example can be found in our open source app [here](https://github.com/revtel/react-native-nfc-rewriter/blob/master/ios/NfcOpenReader/Info.plist), then it should work!

TheKeenestKeen, 2022-04-12:
Thanks so much @whitedogg13. That did the trick. Thanks for bearing through with my confusion. And thanks for creating react-native-nfc-manager.

closed 2022-04-12 by TheKeenestKeen

nfuenmamelt, 2023-10-30:
@TheKeenestKeen how do you solve this? was the TAG?