revtel/react-native-nfc-manager #510: Can i get tag response without button click event?

kojoo112, 2022-04-14:
I want to get tag without button click event.

how can i do it?

please tell me

ShiriNmi1520, 2022-05-05:
[Example code](https://github.com/revtel/react-native-nfc-manager#usage) provided in readme should do the trick 🤔

Helicosaurus-Rex, 2022-05-05:
@ShiriNmi1520 The exemple you linked to uses an onPress event.

I also don't get how you are supposed to read tags without attaching the processing logic onto another UI event. Do I need to use NFCManager.ndefHandler and the likes? I find the exemples a bit vague.

ShiriNmi1520, 2022-05-05:
@Helicosaurus-Rex Just found a PR mention about updating example code that didn't merge #470, I'll dig into it

Helicosaurus-Rex, 2022-05-05:
Thanks! I'll dig into it as well.

ShiriNmi1520, 2022-05-05:
@kojoo112 @Helicosaurus-Rex 
I've created an [example project](https://github.com/ShiriNmi1520/nfc_test/blob/e5aef49082620fecf515309955b09cbb16ddd086/App.js) using expo bare workflow and some code from #470 that reads NFC tag without clicking anything, sees if this helps or need any improvement

Helicosaurus-Rex, 2022-05-05:
Huge thanks! I should be able to make it work from here.

kojoo112, 2022-05-06:
@ShiriNmi1520 THANK YOU SOOOOOOOOO MUCH!!!!

ShiriNmi1520, 2022-05-06:
No problem, let me know if there's another problem and I'll try to figure it out :)

yfs-2000, 2022-05-24:
@ShiriNmi1520  If the setEventListener method is used and nfcVHandler's transceive method is used, a "No tech request available" warning will appear

ShiriNmi1520, 2022-05-24:
> @ShiriNmi1520 If the setEventListener method is used and nfcVHandler's transceive method is used, a "No tech request available" warning will appear

I'll look into it

closed 2022-09-07 by github-actions[bot]