revtel/react-native-nfc-manager #511: usage with freestyle libre sensor

gabrielew, 2022-04-18:
Is it possible to read the freestyle libre sensor?

I've tried to use with this code

```
await NfcManager.requestTechnology(NfcTech.Iso15693IOS);
    const tag = await NfcManager.getTag();
    const handler = NfcManager.iso15693HandlerIOS;
    console.warn('Tag found', tag);
 // Tag found {"icManufacturerCode": 7, "icSerialNumber": [160, 0, 9, 53, 47, 189], "id": "E007A00009352FBD", "tech": "iso15693"}
    const response = (await handler.getSystemInfo(
      Nfc15693RequestFlagIOS.HighDataRate,
    )) as any;
    console.warn('Response found', response);
// Response found {"afi": -1, "blockCount": 244, "blockSize": 8, "dsfid": -1, "icReference": -1}

    const bytes = await handler.readSingleBlock({
      flags: Nfc15693RequestFlagIOS.HighDataRate,
      blockNumber: 0, // from 0 to 255
    });

// [76,76,16,26,3,0,0,0] => how can I covert this to readable glucose values?
```

gui-dos, 2022-04-18:
I am not a React Native developer but I just discovered that this new project https://github.com/ddtch/react-native-libre-manager is successfully reusing my native Swift classes.

gabrielew, 2022-04-18:
I already tried this lib but doesn't work.

I will look through your classes

gabrielew, 2022-04-18:
> I am not a React Native developer but I just discovered that this new project https://github.com/ddtch/react-native-libre-manager is successfully reusing my native Swift classes.

I already tried this lib but doesn't work.

I will look through your classes

meanwhile what number do you use in block number?

or how can I covert this to readable glucose values? [76,76,16,26,3,0,0,0]

gui-dos, 2022-04-18:
The Libre 1 stores the measured glucose and temperature data in 43 FRAM blocks divided in tre sections of 3 + 37 + 3 blocks which are prefixed by a CRC16 checksum. The first two bytes you are scanning are just the CRC of the first section (the “header”) whose fifth byte stands for the sensor status (`03` = ready),

It’s just the beginning of a long story. A very detailed free technical paper on the Libre Internals is to be found on [“Security analysis of a Connected Glucose Sensor for Diabetes”](https://passthesalt.ubicast.tv/protected/videos/v125f57aae7bfmvm2r6cajw3qr9nws/attachments/pts2020_talk_15_pique_curiosity_not_diabetic_fingers_technical_report.pdf).

gabrielew, 2022-04-22:
@gui-dos hey there, which file has the brazil sensor functionality?

closed 2022-09-01 by github-actions[bot]

krazy527, 2022-09-13:
function toHexString(byteArray) {
    return Array.from(byteArray, function (byte) {
      return ('0' + (byte & 0xff).toString(16)).slice(-2);
    });
  }