revtel/react-native-nfc-manager #519: Native error ````TypeError: null is not an object (evaluating 'NativeNfcManager[name]')```

ShiriNmi1520, 2022-05-13:
## Environment
System:
    OS: macOS 12.3.1
    CPU: (8) arm64 Apple M1
    Memory: 90.84 MB / 16.00 GB
    Shell: 5.8 - /bin/zsh
  Binaries:
    Node: 18.0.0 - /var/folders/t2/324ft6l10d3frmgxbdrbrhcr0000gn/T/yarn--1652422822216-0.8163981801209843/node
    Yarn: 1.22.18 - /var/folders/t2/324ft6l10d3frmgxbdrbrhcr0000gn/T/yarn--1652422822216-0.8163981801209843/yarn
    npm: 8.6.0 - /opt/homebrew/bin/npm
    Watchman: 2022.03.21.00 - /opt/homebrew/bin/watchman
  Managers:
    CocoaPods: 1.11.3 - /opt/homebrew/bin/pod
  SDKs:
    iOS SDK:
      Platforms: DriverKit 21.4, iOS 15.4, macOS 12.3, tvOS 15.4, watchOS 8.5
    Android SDK:
      API Levels: 29, 30, 31, 32
      Build Tools: 28.0.3, 29.0.2, 30.0.0, 30.0.2, 30.0.3, 31.0.0, 32.0.0, 32.1.0
      System Images: android-32 | Google APIs ARM 64 v8a, android-32 | Google APIs Intel x86 Atom_64
      Android NDK: Not Found
  IDEs:
    Android Studio: 2021.2 AI-212.5712.43.2112.8512546
    Xcode: 13.3.1/13E500a - /usr/bin/xcodebuild
  Languages:
    Java: 17.0.1 - /usr/bin/javac
  npmPackages:
    @react-native-community/cli: Not Found
    react: 17.0.1 => 17.0.1 
    react-native: 0.64.3 => 0.64.3 
    react-native-macos: Not Found
  npmGlobalPackages:
    *react-native*: Not Found

## react-native-nfc-manager version
3.11.0

## Package.json
```json
"dependencies": {
    "@react-native-community/datetimepicker": "4.0.0",
    "@react-native-masked-view/masked-view": "^0.2.6",
    "@react-native-picker/picker": "2.4.1",
    "@react-navigation/bottom-tabs": "^6.3.1",
    "@react-navigation/native": "^6.0.10",
    "@react-navigation/stack": "^6.2.1",
    "axios": "^0.27.2",
    "eas-cli": "^0.52.0",
    "expo": "^44.0.0",
    "expo-barcode-scanner": "~11.2.0",
    "expo-camera": "~12.1.2",
    "expo-image-manipulator": "~10.2.0",
    "expo-location": "~14.0.1",
    "expo-modules-autolinking": "^0.5.5",
    "expo-secure-store": "~11.1.0",
    "expo-splash-screen": "~0.14.1",
    "expo-status-bar": "~1.2.0",
    "expo-updates": "~0.11.7",
    "i18n-js": "^3.8.0",
    "i18next": "^21.6.16",
    "i18next-http-backend": "^1.2.1",
    "moment": "^2.29.1",
    "react": "17.0.1",
    "react-dom": "18.1.0",
    "react-i18next": "^11.8.13",
    "react-native": "0.64.3",
    "react-native-elements": "^3.0.0-alpha.1",
    "react-native-gesture-handler": "~2.1.0",
    "react-native-keyboard-aware-scroll-view": "^0.9.5",
    "react-native-modal": "^13.0.1",
    "react-native-nfc-manager": "3.11.0",
    "react-native-picker-select": "^8.0.2",
    "react-native-reanimated": "~2.8.0",
    "react-native-safe-area-context": "3.3.2",
    "react-native-screens": "~3.13.1",
    "react-native-table-component": "^1.2.1",
    "react-native-web": "0.17.7"
  },
  "devDependencies": {
    "@babel/core": "^7.12.9",
    "babel-jest": "~28.0.2",
    "jest": "^26.6.3",
    "react-native-clean-project": "^4.0.1",
    "react-test-renderer": "~18.1.0"
  }
```
## To reproduce
* Execute ```expo eject``` command
* Start APP with ```expo start```command
* Error occurs

## Stacktraces
```
[Unhandled promise rejection: TypeError: null is not an object (evaluating 'NativeNfcManager[name]')]
at node_modules/react-native-nfc-manager/src/NativeNfcManager.js:7:39 in callNative
at node_modules/react-native-nfc-manager/src/NfcManager.js:78:2 in NfcManagerBase#unregisterTagEvent
at component/components.js:174:8 in ReadNFC
```

## Excepted behaviour
Successfully implemented NFC feature

## Description
Currently, my app is not working with ```expo start``` command but works with ```expo run:android``` or ```expo run:ios```, perhaps this issue is related to the package version?

whitedogg13, 2022-05-13:
I'm not using `expo`, so I don't really aware the difference between `expo start` and `expo run:android`. But per my understanding, this might be normal because if we don't rebuild the native part (looks like what `expo run:<platform>` does), we don't have the native counterpart for our library, and that's the error is talking about.

Maybe you can check the `expo` doc about whether it rebuild native module and deploy to target while `expo start`?

whitedogg13, 2022-05-13:
BTW, totally off topic but I just find out that you're also from Taiwan! It's good to see another Taiwaness here 🤘 and thanks for your contribution!

ShiriNmi1520, 2022-05-13:
It's good to see you here too and thanks for your effort to make this project so I can complete some magical things :)

About the problem I encountered I'll look into ```expo``` doc to see if I find anything related

closed 2022-08-27 by github-actions[bot]