revtel/react-native-nfc-manager #521: React Native NfcManager.isSupported() returns false

arbnorhaxhiu94, 2022-05-19:
I have installed 'react-native-nfc-manager' package, and have followed all the installation instructions, I have no errors, except that when I run the code below, it returns false.

let isSupported = await NfcManager.isSupported();
console.log('isSupported:', isSupported);

I didn't find any answer on Stackoverflow about this issue and neither in Github, I hope to get a response.

arbnorhaxhiu94, 2022-05-20:
@ShiriNmi1520 thanks for the reply, I am testing it on Samsung Galaxy A30. I still didn't test it on a iOS device.

ShiriNmi1520, 2022-05-23:
@arbnorhaxhiu94 According to [this spec](https://www.gsmarena.com/samsung_galaxy_a30-9579.php), Samsung Galaxy A30 seems don't have the NFC features, did you got another device that is available to test?

arbnor-innos, 2022-05-26:
@ShiriNmi1520 Thank you.

arbnorhaxhiu94, 2022-05-26:
@ShiriNmi1520 Thank you. I tried it on another device, it worked. Seems like all the phones older than 2020 do not support nfc.

ShiriNmi1520, 2022-05-26:
Glad to help :)

closed 2022-05-26 by arbnorhaxhiu94

naheed-shamim, 2025-03-07:
Hi, I am checking
`
const supported = await NfcManager.isSupported();
`

But it is always returning false. 
Tested on iPhone 13 Pro real device as dev build. 
But it supports NFC

silviosampaio, 2025-04-15:
@naheed-shamim I fixed that issue just redoing the 1st and 3rd step on the documentation (about enable NFC capabilities on the App Identifier and Xcode Target)