revtel/react-native-nfc-manager #523: P2P / Phone 2 Phone Transfer.

pat-mw, 2022-05-21:
Hi there,

I have an application in mind which would massively benefit from NFC technology but requires data transfer between two "active" devices rather than just a device and a tag.

We are prototyping Android <-> Android data transfer but ultimately want this to function cross-platform between Android <-> iOS.

The data transfer doesn't need to be complicated. Ultimately we just need the two devices to share an 8-bit code with eachother and this code gets saved into the App memory on both devices: Device 1 sends "12345678" to Device 2 (and gets saved) - and Device 2 sends "87654321" to Device 1 (and gets saved). Is this possible?

I saw an issue opened in 2021 which apparently was resolved with a straight "No"... I'm surprised this isn't a more common use-case given that NFC has been around for 18 years now. Is there absolutely no way to achieve this? Even by writing low-level NFC commands?

Would appreciate some guidance.

Thank you in advance

SurpriseMF3000, 2022-05-27:
I´m having the same issue and also need a solution to interact via NFC between two iPhones.

tylercallison, 2022-08-29:
Apple is releasing this feature in the upcoming iOS. It would be nice if we could get support for P2P NFC support.

https://developer.apple.com/tap-to-pay/
https://developer.apple.com/documentation/ProximityReader?changes=latest_minor

closed 2022-12-15 by github-actions[bot]