revtel/react-native-nfc-manager #524: [iOS] Reader not working on iOS

GSFZamai, 2022-05-27:
Hi, first of all thank you for the Lib.
I´m trying to implement a NFC reader application. 
But it´s not working on iOS (iPhone XR iOS 15), on my android it´s working fine.

I'm running with expo. Followed all of the [Setup's steps](https://github.com/revtel/react-native-nfc-manager/blob/main/setup.md).

Here's the code :

```
async function readNdef() {

        try {
            await NfcManager.start();

            if (Platform.OS === "android") {
                handleModalConfiguration("Aproxime o cartão");
            }

            await NfcManager.requestTechnology(NfcTech.Ndef);

            const tag = await NfcManager.getTag();
            return tag;
        } catch {
            handleModalConfiguration("Falha ao realizar leitura do cartão");
        } finally {
            NfcManager.cancelTechnologyRequest();
        }
    }
```

As I said on Android it´s working fine, but, on iOS all i get is the UI requesting to Tap my NFC Tag.

uguraktas, 2022-07-13:
Is there any solution to this?

optitrac, 2022-07-17:
I'm having the same issue--the system scanning popup does not open on IOS. On Android it works fine.
This code used to work in our project, but when we upgraded from react-native 61 to 68, it stopped working on IOS.

bryce222, 2022-07-22:
Having the same issue, working on Android but not on my IOS 15.5. I think it is the package issue, not updated to the latest IOS, it doesn't support NfcManager.requestTechnology() on IOS, hope the author can get this fix.

whitedogg13, 2022-07-22:
Hi all, sorry for the late response. I’m pretty busy recently. I will look into the issue this weekend!

whitedogg13, 2022-07-23:
Hi all, I have tested our lib on RN `66` / `68`, and all of them work correctly on iOS `15.5`.

I think there maybe two reasons for the condition you're facing:
1. `iOS` configuration: `iOS` does require more configurations than `Android`, please see [here](https://github.com/revtel/react-native-nfc-manager/blob/main/setup.md) in more detail. (I also recorded an YouTube video [here](https://www.youtube.com/watch?v=rAS-DvNUFck)).
2. Some tags won't be recognized by `iOS`. Ya, that's true, some tags just won't be recognized by `iOS core NFC framework` (but sadly they can be recognized by Apple's builtin app such as shortcut, unfair competition), for example `NXP Mifare Classic`.

I'll suggest if the above issues still present, you can try to run our [demo app](https://apps.apple.com/tw/app/nfc-rewriter/id1551243964) to see if it works on your device / tag, and it's also open source [here](https://github.com/revtel/react-native-nfc-rewriter).

GSFZamai, 2022-07-25:
Hello @whitedogg13, I've followed the required configurations and didn't work.
I think, unfortenly, that my case fit on the second condition, considering that I've already tested your demo's app (and some others from the App Store) and did't work too...
Thank you for the response and attention.

closed 2022-07-26 by whitedogg13

pke, 2023-12-04:
@whitedogg13 Demo app does not recognise credit cards nor health insurance cards.