revtel/react-native-nfc-manager #528: isSupported returns true for Galaxy A20s and A50 also if they do not support NFC

27leaves, 2022-06-02:
We have some users now popping up in our crashlytics that experience the following crash:

```
Fatal Exception: android.content.ActivityNotFoundException: No Activity found to handle Intent { act=android.settings.NFC_SETTINGS }
```

Which is after a ` NfcManager.goToNfcSetting()`. The only way this is called is when `NfcManager.isSupported()` is resolved to `true`.

In my little research I found that apparently those phones come in 2 different versions - one with and one without NFC support? Maybe it could be that there the logic somehow fails.

`react-native-nfc-manager` version: 3.11.0
`react-native` version: 0.66.0

Thank you!

whitedogg13, 2022-06-15:
@27leaves thanks for reporting this!

The `goToNfcSetting`'s native implementation tries to launch a standard intent `ACTION_NFC_SETTINGS`:
```java
startActivity(new Intent(Settings.ACTION_NFC_SETTINGS));
```

So I think the issue might be caused by some phone manufactorer customizing this intent.

I think a possible solution might be catching the native exception and resolves to a boolean value to indicate whether the setting's activity can be found. I will fix this one in next release.

whitedogg13, 2022-06-25:
Hi @27leaves this has been fixed in latest version `3.13.5`.

In this version, `goToNfcSetting` will resolve to a `boolean` value to indicate whether we can jump to system NFC setting succesfully.

> Sadly, we cannot handle the wrong `isSupported()` issue because it's caused by the device itself. This API is simply querying native `PackageManager`'s `hasSystemFeature()` for `FEATURE_NFC`, and send the response back.

closed 2022-06-25 by whitedogg13