revtel/react-native-nfc-manager #529: [Question] Is there any difference between requestTechnology and setEventListener

namnh-0652, 2022-06-06:
Is there any difference between these code to read a NFC tag?
```js
NfcManager.setEventListener(NfcEvents.DiscoverTag, (tag) => {
   // TODO with tag
}
```
and 
```js
await NfcManager.requestTechnology([NfcTech.Ndef]);
const tag = await NfcManager.getTag();
// TODO with tag
```

Also, I have a NFC `Mifare Plus` tag, which method I should use to read it?, I use your app https://github.com/revtel/react-native-nfc-rewriter but only android can work, iOS not work.

namnh-0652, 2022-06-07:
@whitedogg13 Sorry to mention, but I really want to listen your advice ASAP.
Thank you verry much!

closed 2022-06-07 by namnh-0652

namnh-0652, 2022-06-08:
@whitedogg13 I tested with 2 NFC tags, both of them were read (I just get the tag Id only)

- My code requests  the techs `requestTechnology([NfcA, IsoDep])`
- (1) NFC (NPX Mifare Plus) contains techs `NfcA, IsoDep` => read ok
- (2) NFC (Mifare Classic) contains techs `NfcA, MifareClassic, NDEF` => read ok

Can my app work with (1) only ?

namnh-0652, 2022-06-09:
@whitedogg13 Thank you very much for your instantly support.
Sadly, the [React Native NFC ReWriter App](https://github.com/revtel/react-native-nfc-rewriter) still not able to read both of my tags above. Only work in android.

whitedogg13, 2022-06-14:
@namnh-0652 thanks for following this. FYI, I have tried `NXP Tag Info` app on `iOS`, and it cannot recognize its own `Mifare Classic` tags as well.