revtel/react-native-nfc-manager #530: Course Demo First Example

edulecca, 2022-06-13:
I'm trying to implement/refactor the course's Game example given on the first chapter

`

    React.useEffect(() => {
        let count = 5;
        NfcManager.setEventListener(NfcEvents.DiscoverTag, (tag) => {
            count--;
            NfcManager.setAlertMessageIOS(`${count}...`);
            if (count <= 0) {
                NfcManager.unregisterTagEvent().catch(() => 0);
                setDuration(new Date().getTime() - start.getTime());
            }
        });

        return () => {
            NfcManager.setEventListener(NfcEvents.DiscoverTag, null);
        };
    }, [start]);
`
but no able to refacotr with the ` await NfcManager.requestTechnology([NfcTech.Ndef])` method suggesed

whitedogg13, 2022-06-14:
Hi @edulecca thanks for your support!

You're right, the `requestTechnology` won't emit `DiscoverTag` event, so that's why you won't be able to receive it.

Per my point, the `Game` logic is designed to demonstrate `registerTagEvent` approach, which is different from `requestTechnology` approach, so try to use `requestTechnology` for refactoring is not suggested.

Thanks!

edulecca, 2022-06-15:
ty @whitedogg13 appreciate your course and your work done

closed 2022-06-15 by edulecca