revtel/react-native-nfc-manager #531: [iOS] Why NFCPollingISO18092 has been rejected?

kikuchy, 2022-06-18:
I tried to use react-native-nfc-manager to read FeliCa tags.
It's awesome library ! Thank you for developting and maintaining.

The example on README.md works file on Android, but not worked on iOS with FeliCa tags.
Tapping the FeliCa tag on the iPhone, the app don't back form iOS's NFC reading modal dialog (with massage "Please tap NFC tags").
It looks `NFCPollingISO18092` is missing on `NFCTagReaderSession`'s `initWithPollingOption:delegate:queue:`. I added `NFCPollingISO18092` for the option, it works.

And I found this commit.
https://github.com/revtel/react-native-nfc-manager/commit/64115f80f8cd93ed5ef3b01987d6c628a5a9127b

Why `NFCPollingISO18092` has been removed? It looks like it's necessary for work with FeliCa tags and it doesn't cause crash.

I'll make a PR to add `NFCPollingISO18092` if it's ok.

whitedogg13, 2022-06-18:
@kikuchy Thanks for reminding this! I have encountered some strange freezing issue while enabling `NFCPollingISO18092` long time ago, so I disabled this flag for a while as a temp solution. BTW, I remember I'm testing with `NTAG 215` at that time.

PR will be great! I'm always looking forward to see new contributor onboarding, and I also hope our library can properly support `Felica`. We should also run more tests to check if the previous issue still present.

kikuchy, 2022-06-19:
Thank you for replying!

Now I see, I'll make a PR later.
I know it's so hard to teasing all kind of physical tags. Thank you for maintaining again.

closed 2022-06-25 by whitedogg13