revtel/react-native-nfc-manager #539: Read the same tag sometime will have diffirent information.

yanwzhang, 2022-07-05:
react-native: 0.64.1
android: 12

When I read one tag, sometime will occur diffrent read result. like this:

[image]
[image]

it is the same id, but sometimes it will miss ndefMessage.

yanwzhang, 2022-07-05:
react-native-nfc-manager: 3.11.0

yanwzhang, 2022-07-06:
when I use NFC reWriter to scan the same NFC tags. 
it's the diffrent result:
[image]
the top screen doesn't have nefMessage and other info. but when I scan again, it will have:
[image]

yanwzhang, 2022-07-18:
> Have the same issue on iOS in our prod app. I'm not able to reproduce it myself but I can see it in our Firebase Crashlytics logs, `ndefMessage` is sometimes not present.
> 
> [image]
> 
> `react`: 0.64.1 `react-native-nfc-manager`: 3.13.2
> 
> EDIT: I am able to consistently reproduce the issue. I initiate the NFC read when I'm already at certain distance of the tag. see the video below:
> 
>  PXL_20220712_145253641.mp4 
> @yanwzhang My solution to fix it for now was to implement the old style use with event listening as shown on the module's page, hope it helps!

it seems to work to implement the old style use with event listening, I'll try more times to test this way, thank you.

yanwzhang, 2022-07-18:
> Have the same issue on iOS in our prod app. I'm not able to reproduce it myself but I can see it in our Firebase Crashlytics logs, `ndefMessage` is sometimes not present.
>
> [image]
>
> `react`: 0.64.1 [...]

I test many times, it will some time occur the same situation, ndefMessage sometimes doesn't have.

closed 2022-11-03 by github-actions[bot]

gigeos, 2023-01-06:
Still occur for me. I tried to read tag with other app like NFC Tools and I reproduce the same situation.
According to my analytics, this happens on both ios 15 and 16

Type: NXP NTAG215
Data format: NFC Type 2