revtel/react-native-nfc-manager #544: How to determine if scanning is canceled and not failed

berenteb, 2022-08-03:
I catch an error while requesting technology, but the error object is empty and can't determine whether the scanning threw an error or the user canceled it (I need a separate outcome).

closed 2022-11-18 by github-actions[bot]

szfdiwang, 2024-10-15:
please, anyone can fix this？

aminhdev, 2024-12-04:
I have the same question

elavneet, 2024-12-04:
Just put your code in try catch. Within catch block do this:

`
  const handleException = (ex) => {
    if (ex instanceof NfcError.UserCancel) {
      // User cancelled 
      
    } else {
      console.warn(ex);

      if (Platform.OS === 'ios') {
        NfcManager.invalidateSessionWithErrorIOS(`${ex}`);
      } else {
        Alert.alert('NFC Error', `${ex}`);
      }
    }
  };
  
catch (error) {
      console.log(`Error`, error);
      handleException(error);
      // if error.message is set, it means it's most likely duplicate registeration error message
      // we can ignore that, and only show the error message when it's actually failed write
      // if (!error.message) setErrorMsg('Failed. Please try again.');
} 
`