revtel/react-native-nfc-manager #546: iOS cancelTechnologyRequest -> requestTechnology -> Duplicate registration

asagalovskiyLB, 2022-08-12:
In iOS, `NfcManager.cancelTechnologyRequest()` does not immediately cancel the request, and has no feedback event about when the request has been canceled. If you execute  `NfcManager.requestTechnology()` in the meantime, it will return with an error: `Oops! [Error: Duplicated registration]`

How to reproduce:
1. Activate the scan with `NfcManager.requestTechnology()`
2. Scan the tag
3. Immediately swipe down on the iOS's scan prompt screen to hide it.
4. Activate the scan again with `NfcManager.requestTechnology()`

wordisthebird, 2022-10-21:
Any update on this? or how to handle error?

erikostling, 2023-01-05:
The only solution that I've found to this so far is delaying the read requestTechnology call (2 seconds). Seems not like an ideal solution.

closed 2023-04-21 by github-actions[bot]

dcanora, 2023-06-17:
Has anyone found a solution to this? I believe it may help me solve my issue... I experienced this duplicate registration problem, so I tried to call requestTechnology() once, then loop through calling getTag() multiple times, and finally cancelTechnology() when the user is done but I'm getting the issue outline in #645.