revtel/react-native-nfc-manager #547: readNdef function doesn't contains NDEF Message.

SurpriseMF3000, 2022-08-14:
Hello,
im calling the `readNdef()` function successful and detect a Tag but the Tag doesn't contains a NDEF message. (It does)

```
  async function readNdef() {
    try {
      // register for the NFC tag with NDEF in it
      await NfcManager.requestTechnology(NfcTech.Ndef);
      // the resolved tag object will contain `ndefMessage` property
      const tag = await NfcManager.getTag();
      console.warn('Tag found', tag);
    } catch (ex) {
      console.warn('Oops!', ex);
    } finally {
      // stop the nfc scanning
      NfcManager.cancelTechnologyRequest();
    }
  }
```

Tag:
`{"canMakeReadOnly": true, "id": "04EA5002F06E80", "isWritable": true, "maxSize": 492, "ndefMessage": [{"id": "", "payload": [Array], "tnf": 1, "type": [Array]}], "techTypes": ["android.nfc.tech.NfcA", "android.nfc.tech.MifareUltralight", "android.nfc.tech.Ndef"], "type": "NFC Forum Type 2"}`

Thanks for Help;)

SurpriseMF3000, 2022-08-17:
I figured out that the `tag.ndefMessage` encryption is `ASSCI`.

closed 2022-08-21 by SurpriseMF3000

chattago2002, 2023-06-01:
> I figured out that the `tag.ndefMessage` encryption is `ASSCI`.

Hi. Have you solved? I'm trying to get the real message but I can't.

SurpriseMF3000, 2023-06-05:
Hey, you just need to decrypt the ASSCI encrypted payload.
You can use decryption libraries/ packages for that.