revtel/react-native-nfc-manager #559: Support for iOS ProximityReader

tylercallison, 2022-09-21:
With iOS 16, Apple released the [ProximityReader API](https://developer.apple.com/documentation/ProximityReader). This could be used for a variety of uses such as iPhone to iPhone NFC data transfer. 

Under the secure layer of Apple Pay, this could be a major feature to enable developers to create secure data transfers using NFC.

More info:
https://developer.apple.com/tap-to-pay/
https://developer.apple.com/documentation/ProximityReader

whitedogg13, 2022-09-24:
@tylercallison thanks for the infomation! I'll definitely take a look!

tylercallison, 2022-10-12:
@whitedogg13 Do you think this would be something we might see implemented within this library?

closed 2023-04-27 by github-actions[bot]