revtel/react-native-nfc-manager #560: Communication between 2 phones

cfontana0, 2022-09-22:
Hi Richie, thanks for your hard work here.

Do you know if it's possible to interact between 2 phones using NDEF and this library?

I understand that the HCE feature is not allowed on iPhones, right? That is possible only on Android.

However, we're able to transmit information from iPhones and Androids using NDEF and based on it, we're able to record information on NFC cards. Could we use the same technology to interact between 2 phones? Ideally, using React Native, but if not, it can be using native code.

whitedogg13, 2022-09-24:
NDEF itself is only a format to store data into NFC tags. The primary difficulty to achieve inter-devices communication is due to the device's NFC role. 

To exchange data between devices using NFC, there're two conditions:
- NFC reader to NFC tag/card
- NFC peer to NFC peer (P2P)

The OS support for above features look like this:

|   |     Reader      |  Tag/Card |  P2P |
|----------|:-------------:|------:|------:|
| iOS |  yes | no | no |
| Android |   yes   | yes(hce) | deprecated |

By looking at this table, we can conclude:
- iOS to iOS: cannot communicate to each other, because they're both **reader**s (two **readers** cannot talk to each other!)
- iOS to Android: possible, iOS as **readers** and Android as **emulated tags via HCE**
- Android to Android: possible, both can be **readers** or **emulated tags**, but two devices **must** have different roles.

However, even the second / third options are possible, it's not trivial to implement because you will need an upper layer control potocol to determine the device role, especially considering Android-to-Android.

Hope that helps!

cfontana0, 2022-11-01:
Hey @whitedogg13 thanks for your message. I agree with you! 
Even the Android-to-Android option is a bit complicated, since it does not work for all phones, there are restrictions according to the brand and model, right?

closed 2023-02-15 by github-actions[bot]

nick0504k, 2024-01-30:
https://developer.apple.com/support/hce-payment-transactions-in-payment-apps/

Starting from iOS 17.4, Apple introduced new APIs for developers to support contactless payment transactions from within their banking or wallet apps using HCE, but only for users based in the European Economic Area (EEA)

----
can try android to android with this library 
https://github.com/appidea/react-native-hce

Henry-E, 2024-06-26:
> NDEF itself is only a format to store data into NFC tags. The primary difficulty to achieve inter-devices communication is due to the device's NFC role.
>
> To exchange data between devices using NFC, [...]

Like this table is really helpful and should be easier find.