revtel/react-native-nfc-manager #565: NFC IS NOT SUPPORTED ON PAX A930

NamComputer, 2022-10-01:
**ISSUE**
 I'm currently using 'react-native-nfc-manager' to access my physical PAX A930 which supported nfc. I did follow up the installing doc of the packages BUT when i used the _NfcManager.isSupported()_ function. It returned false. I don't know what's wrong. Could somebody help me? Thanks 

**CODE**
```
`import React, { useState, useEffect } from 'react';
import {
  SafeAreaView,
  ScrollView,
  StatusBar,
  StyleSheet,
  Text,
  View,
  TouchableOpacity
} from 'react-native';
import NfcManager, {NfcEvents} from 'react-native-nfc-manager';

export default App = ()=> {
  const [hasNfc, setHasNFC ] = useState(null);

  //This function check the device's hardware which can be able to read the nfc or not
  useEffect(() => {
    const checkIsSupported = async () => {
      const deviceIsSupported = await NfcManager.isSupported()
      console.log('device is supported:',deviceIsSupported)
      setHasNFC(deviceIsSupported)
      if (deviceIsSupported) {
        await NfcManager.start()
        console.log("NFC scan event is starting...")
      }
    }

    checkIsSupported()
  }, [])

  //Let’s write the code that will read an NFC tag:
  useEffect(() => {
    NfcManager.setEventListener(NfcEvents.DiscoverTag, (tag) => {
      console.log('tag found')
    })

    return () => {
      NfcManager.setEventListener(NfcEvents.DiscoverTag, null);
    }
  }, [])

  const readTag = async () => {
    await NfcManager.registerTagEvent();
  }
  const cancelReadTag = async () => {
    await NfcManager.unregisterTagEvent();
  }

  if (hasNfc === null) return null;
  return(
  !hasNfc ?
        <View style={styles.backgroundStyle}>
          <Text>NFC not supported</Text>
        </View>
    :
      <SafeAreaView style={styles.sectionContainer}>
        <TouchableOpacity style={[styles.btn, styles.btnScan]} onPress={readTag}>
          <Text style={{ color: "white" }}>Scan Tag</Text>
        </TouchableOpacity>
        
        <TouchableOpacity style={[styles.btn, styles.btnCancel]} onPress={cancelReadTag}>
          <Text style={{ color: "white" }}>Cancel Scan</Text>
        </TouchableOpacity>
      </SafeAreaView>
  ) 
};

const styles = StyleSheet.create({
  backgroundStyle:{
    flex:1
  },
  sectionContainer:{
    flex:1,
    flexDirection:'row',
    marginVertical:100,
    justifyContent: 'center',
    textAlign:'center'
  },
  btn:{

  },
  btnScan:{
    backgroundColor:"blue"
  },
  btnCancel:{
    backgroundColor:"red"
  }
});

`
```

NamComputer, 2022-10-03:
Can you pls help me @whitedogg13?

LeHoaiVu-1997, 2022-10-31:
@NamComputer You can't use react-native-nfc-manager or android.nfc on PAX machines as I remember. In order to use nfc reader on those pos terminals, you must use a Java library called Neptune lite api and make your own React methods. You can find it here: https://docs.hips.com/docs/pax-a920

app3-speedPOS, 2022-10-31:
Thanks mate @LeHoaiVu-1997. At that time, I also tried to request my quesiton to Pax's team and It is what it is ... Like you said above :))

NamComputer, 2022-10-31:
Thanks mate @LeHoaiVu-1997. At that time, I also tried to request my quesiton to Pax's team and It is what it is ... Like you said above :))

closed 2023-02-14 by github-actions[bot]

Juliusjhn, 2023-02-28:
how to make your own React methods? and call java function

hshah965, 2024-02-06:
@NamComputer did you figure this out? I'm trying to achieve the same thing i.e access NFC on PAX A910 device via react native.

Juliusjhn, 2024-02-06:
Since there might not be a ready-made React Native library for the Neptune Lite API, you may need to create your own native modules. This involves writing Java (Android) code to interact with the Neptune Lite API and exposing methods that can be called from your React Native JavaScript code. Implement the necessary logic in your React Native components to trigger the native methods when NFC events occur. You might use the React Native Event Emitter or Promises to communicate between your native code and JavaScript code.