revtel/react-native-nfc-manager #568: MiFareIOS - Slow and Inconsistent Performance

ntbutler-nbcs, 2022-10-10:
Hi all.

I'm experiencing some odd performance issues using this on my iPhone 12 Pro (iOS 16.0). I am trying to scan some test cards and it is taking sometimes up to 30sec for me to get the tag ID. However if I open an app such as NFC Tools, the cards scan immediately.

The cards I'm scanning are both Mifare (ISO 14443-3A), one appears in NSF tools as a "NXP - Mifare DESFire EV1 4k" and the other appears as "NXP - Mifare Ultralight EV1 48 bytes". Both cards scan within a second in that app.

When I try with the code below, I am seeing some very long delays between each of the console.log steps, particularly between step 3 and 4.

```javascript
console.log("Step 1");
let resp = await NfcManager.requestTechnology(NfcTech.MifareIOS, {
    alertMessage: "Scan a registered Card"
});

console.log("Step 2");
NfcManager.cancelTechnologyRequest(resp);

console.log("Step 3");
let tag = await NfcManager.getTag();

console.log("Step 4 - tag: " + tag.id);
actionCard(tag.id);
```

My specific process here is:

1. Launch the app and trigger the code above via a button press. The iPhone pops up with the "Ready to Scan" popup as usual. The XCode output log displays "NFCTag didBecomeActive"
2. Bring the card near the phone. The XCode output log displays "NFCTag didDetectTags" immediately.
3. Anywhere from 3-20 seconds later, I see "Step 2" logged, the technologyrequest is cancelled, and I see "Step 3" logged.
4. Anywhere from 5-30 seconds later again, I see "Step 4 - tag: <tagid>" logged, and my next function is triggered.

Is there something that might be causing the delays? I am running version 3.14.0 of this library, and I can also get this to behave the same way on an iPhone 8 Plus

closed 2023-01-24 by github-actions[bot]

hkaras19, 2023-07-24:
@ntbutler-nbcs were you able to figure this out? I am running into the same issue.

ntbutler-nbcs, 2023-07-24:
@hkaras19 unfortunately not sorry - for the project I was working on, I had to abandon iOS MiFARE and go with an external USB card reader