revtel/react-native-nfc-manager #578: Keep reading tags

robertcoroianu, 2022-10-31:
Hello,
      It's there a way to keep scanning the tags not only after  a button was press?

robertcoroianu, 2022-10-31:
I added a event, for discoverTags and not unregister :)
```
`  function readNdef() {
    NfcManager.setEventListener(NfcEvents.DiscoverTag, (tag: any) => {
      let tagFound = tag;
      console.log(tagFound);
      // NfcManager.unregisterTagEvent();
    });
    NfcManager.registerTagEvent();
  }`
```

closed 2022-10-31 by robertcoroianu

Productivix, 2023-06-15:
hi, can you @robertcoroianu  please the complete code example ? thank you !

robertcoroianu, 2023-07-11:
> hi, can you @robertcoroianu please the complete code example ? thank you !

```
import {View, StyleSheet, Text} from 'react-native';
import React, {useEffect} from 'react';
import NfcManager, {NfcEvents} from 'react-native-nfc-manager';

NfcManager.start();

function RfidReader(props: {sendAuth: (code: string, type: number) => void}) {
  const changeEndianness = (value: string) => {
    const result = [];
    let len = value.length - 2;
    while (len >= 0) {
      result.push(value.substring(len, 3));
      len -= 2;
    }
    return result.join('');
  };
  function barcodeRecognized(code: any): void {
    code = changeEndianness(code);
    props.sendAuth(code, 2);
  }
  async function readNdef() {
    if (await NfcManager.isSupported()) {
      NfcManager.setEventListener(NfcEvents.DiscoverTag, (tag: any) => {
        let tagFound = tag;
        console.log(tagFound);
        barcodeRecognized(tag.id);
        // NfcManager.unregisterTagEvent();
      });
      NfcManager.registerTagEvent();
    } else if (await NfcManager.isEnabled()) {
      console.log('NFC Not Active');
    }
  }

  // Run once
  useEffect(() => {
    readNdef();
  }, []);
  return ( );
}

export default RfidReader;
```

Productivix, 2023-07-11:
yes ! thank you :  it runs very well !