revtel/react-native-nfc-manager #581: Random application crashes due to: Illegal callback invocation from native module...

Sixzero, 2022-11-07:
The raw stacktrace we receive in our application countless times:
```
java.lang.RuntimeException: Illegal callback invocation from native module. This callback type only permits a single invocation from native code.
        at com.facebook.react.bridge.CallbackImpl.invoke(CallbackImpl:26)
        at community.revteltech.nfc.NfcManager.parseNfcIntent(NfcManager:1341)
        at community.revteltech.nfc.NfcManager.onNewIntent(NfcManager:1297)
        at com.facebook.react.bridge.ReactContext.onNewIntent(ReactContext:277)
        at com.facebook.react.ReactInstanceManager.onNewIntent(ReactInstanceManager:553)
        at com.facebook.react.ReactActivityDelegate.onNewIntent(ReactActivityDelegate:146)
        at com.facebook.react.ReactActivity.onNewIntent(ReactActivity:102)
        at android.app.Activity.performNewIntent(Activity.java:8306)
        at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1429)
        at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1442)
        at android.app.ActivityThread.deliverNewIntents(ActivityThread.java:4241)
        at android.app.ActivityThread.handleNewIntent(ActivityThread.java:4248)
        at android.app.servertransaction.NewIntentItem.execute(NewIntentItem.java:56)
        at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
        at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2438)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loopOnce(Looper.java:226)
        at android.os.Looper.loop(Looper.java:313)
        at android.app.ActivityThread.main(ActivityThread.java:8669)
        at java.lang.reflect.Method.invoke(Method.java:-2)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1135)
```

We are on version:
`"react-native-nfc-manager": "3.13.5",`

I don't think `.requestTechnology` is called twice, and also.
On line `NfcManager:1341` line there is this line:
`techRequest.getPendingCallback().invoke(null, null);`

What interests us is that things were good for years, and now suddenly, things broke in the last weeks and crashing the application for many. We thought it is worth starting an issue about it. 

What do you think, what could be the cause?

crzycoder, 2022-11-09:
Any progress ???
same issue facing

Sixzero, 2022-11-09:
We are thinking about removing the package since we have no hard reliance on NFC. Can1t do better for now.

whitedogg13, 2022-11-15:
Hi all, sorry for the late response. I will try to investigate this. Can you tell me in which RN / Android version this happened?