revtel/react-native-nfc-manager #598: Discrepancy between Android and iOS getTag result

erikostling, 2022-12-13:
I'm getting different results when calling getTag on Android vs iOS. Is there any obvious reason behind this? Issue seems related/identical to #450.

Code:

```
await NfcManager.requestTechnology(NfcTech.Ndef)
const tag = await NfcManager.getTag()
```

Android:

```
{
  "canMakeReadOnly":true,
  "id":"046C70F2CB7281",
  "isWritable":true,
  "maxSize":137,
  "ndefMessage":[
    ...
  ],
  "techTypes":[
     "android.nfc.tech.NfcA",
     "android.nfc.tech.MifareUltralight",
     "android.nfc.tech.Ndef"
  ],
  "type":"NFC Forum Type 2"
}
```

iOS:

```
{
  "id":"046C70F2CB7281",
  "ndefMessage":[
    ...
  ],
  "tech":"mifare"
}
```

Expected result:

```
getTag: () => Promise<TagEvent | null>

...

export interface TagEvent {
  ndefMessage: NdefRecord[];
  maxSize?: number;
  type?: string;
  techTypes?: string[];
  id?: string;
}
```

Debug info:

```
react-native: 0.70.2
react-native-nfc-manager: 3.14.1
```

whitedogg13, 2022-12-20:
Hi @erikostling great question! 

Regarding the ugly type `TagEvent`, it's mainly because:
- `iOS` and `Android` handle things differently
  - they report different fields from native layer
  - some fields even requires extra API calls
- different NFC technologies have different fields

So rather than putting efforts to make all fields identical in both OS, I only focused on following ones (if they present):
- `id`: means the NFC tag's physical identifier
- `ndefMessage`: the NDEF contained by scanned NFC tag

It's simply because In my perspective, `id` and `ndefMessage` are the most important fields for most use cases.

closed 2023-04-06 by github-actions[bot]

ombogdanJoinToIt, 2023-08-17:
why can't I get the ID in the card when I read it from iOS?

ombogdanJoinToIt, 2023-09-09:
how to get a tag id on ios. an empty array is now returned. although everything is fine on Android

yulolimum, 2025-01-12:
> why can't I get the ID in the card when I read it from iOS?

@ombogdanJoinToIt did you ever figure this out? no ID is returned on iOS.