revtel/react-native-nfc-manager #604: Data is clearing while trying to write to the NFC

pruthvirajkarurcodewave, 2023-01-03:
Hi Team,

Data in the NFC is getting erased. Our product is already live and erase issues are coming which is very alarming for the business. 
Please help us out in resolving in this issue.

I am using the following code for writing please suggest.
 

` 
 const writeTestFinal = async text => {

    var result = false;

    let type = 'TEXT';

    try {

      var value = 'test data to write';

      await NfcManager.requestTechnology(NfcTech.Ndef, {
        alertMessage: 'Ready to write some NDEF',
      });

      let bytes = null;
      if (type === 'TEXT') {
        bytes = Ndef.encodeMessage([Ndef.textRecord(value)]);
      } else if (type === 'URI') {
        bytes = Ndef.encodeMessage([Ndef.uriRecord(value)]);
      }

      if (bytes) {
        await NfcManager.ndefHandler.writeNdefMessage(bytes);

        result = true;
      }
    } catch (ex) {
      console.warn('ERROR IN WRITETESTFINAL : ', ex);
    } finally {
      NfcManager.cancelTechnologyRequest();
    }

    return result;
  };`
Happy flow works: If the users just taps just once then the data write successfully.

Real world scenario: If the user tries to move the nfc fast to make the transfer fast and moves the nfc in a circular format attached to the device then the NFC tag is getting cleared after this line `await NfcManager.ndefHandler.writeNdefMessage(bytes);
`
I tried with the existing application from the playstore /appstore **_NFC Rewriter_** in that app also i am getting this issue of data clearance . Please suggest how to resolve this.

pruthvirajkarurcodewave, 2023-01-09:
@whitedogg13  Any idea what could be happening here ? Could you please check how to resolve this.

closed 2023-04-25 by github-actions[bot]

thechosenweak, 2024-04-02:
I encountered same issue, how to fix?