revtel/react-native-nfc-manager #620: Using makeReadOnly crashes the app and makes NFC disabled until phone restart

daaain, 2023-03-24:
When using `makeReadOnly` the app crashes (there's a `java.io.IOException` in Sentry that might be due to this) to the point that the UI is completely frozen, the JS error handler doesn't get called and the NFC functionality is completely disabled on the entire Android OS level (other apps can't read NFC any more) until a hard reboot.

So it's almost surely something not quite in the native layer and after most of the day spent debugging I thought I'd try in the reference app and it's the same! Attaching screenshots of the tag below, but it's an NDef one with a custom media type that otherwise has been working for us for years.

Do you know what it could be or how could I diagnose it?

[image]
[image]

MattEndstate, 2023-05-04:
Running into the same issue here on a Pixel 6a running Android 13 and version `3.14.1`.
I have an older Pixel 5a running Android 11 that is not affected by this issue.

daaain, 2023-08-03:
Please remove stale label, it would be great if this error was handled more gracefully in the library so there's no crash.

closed 2023-11-18 by github-actions[bot]

Mehly, 2024-02-19:
I'm facing the very same issue.  NFC becomes device-globally unusable until device-restart.

aminhdev, 2024-11-28:
I got the same problem