revtel/react-native-nfc-manager #623: NfcManager.requestTechnology not working in android 13

Stembotech, 2023-04-01:
NfcManager.requestTechnology not working in android 13

erdenebayrd, 2023-04-02:
Same as me on Iphone SE second edition, But the setEventListener on Discover tag and registerTagEvent method is working properly.

khanrafay, 2023-04-07:
Same here, the application is working fine till Android 11.
on Android 12 and 13 it works with a delay.

I have created a which scans a mifare ultra card and sign in the user.
but in Android 12/13 it scans and then I have to wait 10 seconds till it can scan again.

Tried every possible way, cancelTechnology, etc.

But nothing is working.

OfflinePing, 2023-04-15:
actually same here,
Android 13
Samsung Galaxy S21 FE 5G
and I just cant get it working properly

closed 2023-07-31 by github-actions[bot]