revtel/react-native-nfc-manager #626: Background NFC opens duplicate react native app on Android

abdullahkhalil2, 2023-04-18:
I am using react native nfc manager. When I scan nfc in the background, my app is opened and nfc is handled but I can see in the multitasking that two apps have opened, when with the nfc native handler icon and the other with my app handler. 
Because of this behavior of the app is weird. Mutiple success popups would appear and multiple navigation would occur. Seems like changing something on one app triggers action on the other app. below is the android manifest I use to open background tags. On IOS the flow is working perfect. For reference we are using the NFC NTAG216, and I have already enabled forgroundEnable mode in the nodemodules nfcManager.java. Also the behaviour is only odd when the app is in the background when scanning tag.
`
        <activity
            android:name="com.holguinmedia.contactapp.MainActivity"
            android:configChanges="keyboard|keyboardHidden|orientation|screenSize|uiMode"
            android:exported="true"
            android:label="@string/app_name"
            android:launchMode="singleTop"
            android:screenOrientation="portrait"
            android:windowSoftInputMode="adjustResize">
            

<intent-filter>
            <action android:name="android.intent.action.MAIN" />
            <category android:name="android.intent.category.LAUNCHER" />
        </intent-filter>

            <intent-filter>
                <action android:name="android.nfc.action.NDEF_DISCOVERED"/>
                <category android:name="android.intent.category.DEFAULT"/>
                <data android:scheme="https"
                    android:host="nfc-frontend-testing.crunchyapps.com" />
            </intent-filter>
            <intent-filter>
                <action android:name="android.nfc.action.NDEF_DISCOVERED"/>
                <category android:name="android.intent.category.DEFAULT"/>
                <data android:scheme="https"
                    android:host="nfc-frontend-testing.crunchyapps.com/nfc/*" />
            </intent-filter>
            <intent-filter>
                <action android:name="android.nfc.action.TECH_DISCOVERED"/>
            </intent-filter>
            <meta-data android:name="android.nfc.action.TECH_DISCOVERED"
                android:resource="@xml/nfc_filter" />

        <intent-filter>
            <category android:name="android.intent.category.DEFAULT" />
            <category android:name="android.intent.category.BROWSABLE" />
            <data android:scheme="com.holguinmedia.contactapp"
                  android:host="*" />
        </intent-filter>
        <intent-filter>
            <category android:name="android.intent.category.LAUNCHER" />
            <category android:name="android.intent.category.BROWSABLE" />
            <data android:scheme="com.holguinmedia.contactapp"
                  android:host="*" />
        </intent-filter>`
[image]
In the photo as it can be seen, the left app is our app and the right app is the system native app handler.

closed 2023-08-03 by github-actions[bot]

ailtonaires, 2024-05-08:
any solution?

abdullahkhalil2, 2024-05-08:
Hi @ailtonaires as far as I remeber I think I added the following line in the manifest file which fixed the issue. 
android:launchMode="singleTask"