revtel/react-native-nfc-manager #629: Android 13 App Launch - java.lang.SecurityException (Tag is out of date)

n-kulic, 2023-04-25:
## Quick Information
- **Physical device: **yes**** 
- **react-native: **0.71.6****
- **react-native-nfc-manager: **3.14.2****

## What Happened?
On `[NDEF_DISCOVERED]((https://developer.android.com/reference/android/nfc/NfcAdapter#ACTION_NDEF_DISCOVERED))` app launch activity app crashes with the following exception:
```
Exception java.lang.SecurityException:
  at android.nfc.Tag.getTagService (Tag.java:381)
  at android.nfc.tech.Ndef.canMakeReadOnly (Ndef.java:344)
  at community.revteltech.nfc.Util.ndefToJSON (Util.java:42)
  ...
```

Related issues:
- NFC unavailable when camera is open - https://stackoverflow.com/a/34678864
- usage of camera blocks NFC reading/writing - https://stackoverflow.com/questions/68636728/usage-of-camera-blocks-nfc-reading-writing
- NFC Not Working When Camera Enabled - https://community.oneplus.com/thread/1551310

## Expected result
App starts without crash/exception

## Steps to reproduce

1. Clone https://github.com/revtel/react-native-nfc-rewriter and start app
2. Terminate app (swipe out) and scan nfc tag with URI record https://washow.netlify.app

## Stacktrace
```
Exception java.lang.SecurityException:
  at android.nfc.Tag.getTagService (Tag.java:381)
  at android.nfc.tech.Ndef.canMakeReadOnly (Ndef.java:344)
  at community.revteltech.nfc.Util.ndefToJSON (Util.java:42)
  at community.revteltech.nfc.NfcManager.buildNdefJSON (NfcManager.java:1389)
  at community.revteltech.nfc.NfcManager.ndef2React (NfcManager.java:1381)
  at community.revteltech.nfc.NfcManager.parseNfcIntent (NfcManager.java:1351)
  at community.revteltech.nfc.NfcManager.start (NfcManager.java:995)
  at java.lang.reflect.Method.invoke
  at com.facebook.react.bridge.JavaMethodWrapper.invoke (JavaMethodWrapper.java:372)
  at com.facebook.react.bridge.JavaModuleWrapper.invoke (JavaModuleWrapper.java:188)
  at com.facebook.jni.NativeRunnable.run
  at android.os.Handler.handleCallback (Handler.java:942)
  at android.os.Handler.dispatchMessage (Handler.java:99)
  at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage (MessageQueueThreadHandler.java:27)
  at android.os.Looper.loopOnce (Looper.java:201)
  at android.os.Looper.loop (Looper.java:288)
  at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run (MessageQueueThreadImpl.java:228)
  at java.lang.Thread.run (Thread.java:1012)
```

closed 2023-08-28 by github-actions[bot]

hkaras19, 2023-10-29:
@n-kulic were you able to find a solution for this? I am currently running into it as well.

rpopovych, 2023-10-31:
@hkaras19 I have not found a solution unfortunately. We've put the implementation on pause for other reasons.

n-kulic, 2023-11-02:
> @rpopovych @n-kulic were you able to find a solution for this? I am currently running into it as well.

@hkaras19 @rpopovych `SecurityException` is thrown when trying to check whether a tag can be made read-only.
Since our app did not required this data we've commented out that part and patched a package.

If your app does not need this data you can as well patch this package:
1. Install and set-up https://www.npmjs.com/package/patch-package
2. Go to `node_modules/react-native-nfc-manager/android/src/main/java/community/revteltech/nfc/Util.java` and comment out [lines 41 to 45](https://github.com/revtel/react-native-nfc-manager/blob/4adccfe2c69cd4c08a1a7c54400049e4f0bc1cc3/android/src/main/java/community/revteltech/nfc/Util.java#L41-L45):
[image]
3. Execute the `npx patch-package react-native-nfc-manager` command after removing the code that causes an exception (or if using yarn, `yarn patch-package react-native-nfc-manager`).
4. After removing code that throws exception run `npx patch-package react-native-nfc-manager`