revtel/react-native-nfc-manager #632: EMV Credit/Debit Card Read

cwtaras, 2023-05-05:
Hello,
I want to add NFC payment to my app but I couldn't find a solution. My app is targeting the merchants and I want to make it possible for them get payments from their clients through my app. Is it possible to do it with this package, if so, how can i implement it? Thanks in advance.

diklou, 2023-05-06:
I'm also interested about this

Noitidart, 2023-07-06:
Anyone able to figure out how to do this? Android and/or iOS? Stripe makes it sound like iOS can now do this as well - https://stripe.com/en-au/terminal/tap-to-pay

[image]

closed 2023-10-18 by cwtaras