revtel/react-native-nfc-manager #633: Reading in a loop

vraja-ids, 2023-05-18:
Hi,

First off thank you so much for this great library! 

Trying to implement scanning bands in a loop for an event so that we can save time in going back and forth in screens. Here is the sample code that we are trying - scanning seems to go fine in a loop, however I cant seem to get out of the loop :

    useEffect(() => {
      const readTag = async () => {
        try {
            NfcManager.cancelTechnologyRequest()
          const newtag = await NfcProxy.readTag();
          if (newtag) {
            const currentMeal = getCurrMeal();
            setCurrentTag(newtag);
            await fetchMealDetails(newtag, currentMeal);
          }
          if (!myError && isContinuousRead && !isLoading) {
            readTag(); // Continue reading if the flag is true
          }
        } catch (ex) {
          console.warn(ex);
          alert('ERROR', 'fail to read tag', [{ text: 'OK' }]);
        }
      };

      if (isContinuousRead && !myError && !isLoading) {
        alert(isContinuousRead);
        readTag();
      }

      return () => {
        setContinuousRead(false); // Cleanup: Set continuous read state to false
      };
    }, [currMeal, myError, isContinuousRead, isLoading]);

2023-05-18 17:46:30.277980-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.278206-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.278464-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.278713-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.278932-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.279182-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.279436-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.279652-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.279886-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.280127-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.280321-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.280543-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.280759-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.280953-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.281188-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.281404-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.281647-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.281900-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.282121-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.282359-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.282574-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.282814-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.283030-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.283254-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.283455-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.283695-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.283933-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.284151-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.284382-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.284598-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.284816-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.285051-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.285284-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.285508-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.285728-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
[... 88 lines cut ...]
2023-05-18 17:46:30.309599-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.309861-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.310101-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.310342-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.310559-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.310769-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.311082-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.311324-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.311558-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.311779-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.311978-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.312197-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.312442-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.312672-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.312909-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.313130-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.313329-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.313535-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.313746-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.313971-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.314201-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.314446-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.314651-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.314840-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]
2023-05-18 17:46:30.315179-0500 NfcOpenReader[11150:5686753] [javascript] [Error: Duplicated registration]

Productivix, 2023-06-15:
hi, did you have a solution ?

valery-lavrik, 2023-08-15:
did you have a solution ?

Productivix, 2023-08-17:
yes , what i rested and adapted with success was : https://github.com/revtel/react-native-nfc-manager/issues/578

closed 2023-12-01 by github-actions[bot]