revtel/react-native-nfc-manager #645: Second call to .getTag() returns previous tag's value

dcanora, 2023-06-16:
I need to allow the user to read multiple NFC cards, with the app processing the tag value read and providing feedback to the user between each read. I have no problem doing this on Android: I call to `.requestTechnology()`, `.getTag()`, `.mifareUltralightHandlerAndroid.mifareUltralightReadPages()` and `.cancelTechnologyRequest()`. I then process the value returned, give the user feedback and then repeat the process to read the next card, until canceled by the user.

This doesn't work on iOS. I've separated the calls to `.requestTechnology(NfcTech.MifareIOS)` and `.cancelTechnologyRequest()` so they only get called before the first call and when canceled, respectively.  I'm trying to use `readNfc()`, below, to do each nfc read.  The first tag reads fine, but when the app calls `readNfc()` a second time - even though no tag is near the phone - the app behaves like the first tag is still present -  `.getTag()` always immediately returns the first card's data (as shown in a console.log of nfcTag.id) and the call to `.sendMifareCommandIOS()` always fails, presumably because there is actually no tag to read. No matter what I do, it seems the first card read is somehow still "seen" by `.getTag()` the second time `readNfc()` is called.

Without fully closing the nfc reader, is there a way to get `.getTag()` to 'move on' from the last tag and successfully loop through reads as I've described here?

      `const readNfc = async () => {
          let readPages = null;
          console.log('[nfcServices] readNfc Starting...');
          
            try {
                // Wait for tag    
                await NfcManager.setAlertMessageIOS('Scan a card…’);
                const nfcTag = await NfcManager.getTag();
                console.log('[nfcServices] readNfc getTag returned ', nfcTag.id);
                
                // Read one four-page array from tag
                readPages = await NfcManager.sendMifareCommandIOS([0x30, 16]);
                await NfcManager.setAlertMessageIOS(‘Card read!');
                console.log('[nfcServices] readNfc readPages: ', readPages);
            
            } catch (ex) {
                // Error
                console.log('[nfcServices] readNfc ERROR: ', ex);
                NfcManager.cancelTechnologyRequest().catch(() => 0);
            } finally {
                console.log('[nfcServices] readNfc returning: ', readPages);
                return(readPages);
            }
      }`

dcanora, 2023-06-16:
BTW, I'm using v3.14.5, and testing on iPhone 8 Plus & 13, iOS 16.5

dcanora, 2023-06-17:
Trying another approach, but also getting stuck... Is there a way to use `NfcManager.setEventListener()` to keep reading non-ndef mifare cards, similar to how it can read multiple ndef encoded cards? 

In my tests, mifare cards without NDEF aren't even recognized by `NfcManager.setEventListener(NfcEvents.DiscoverTag, (tag) => {...}`.

My end goal is to be able to repeatedly read cards (using `NfcManager.sendMifareCommandIOS()` to get data) and process each until the user cancels.  Any advice much appreciated!

ombogdanJoinToIt, 2023-09-09:
how to get a tag id on ios. an empty array is now returned. although everything is fine on Android

closed 2023-12-24 by github-actions[bot]

dcanora, 2024-04-09:
Hello! Checking back in to see if anyone has had any luck reading multiple (not NDEF) tags? Thanks!