revtel/react-native-nfc-manager #651: I had a problem sending the 0xB3 custom command using the handler.customCommand function.

ramazanerdogan, 2023-07-24:
Hello!

I'm using a build of type st25dv64k

For iOS operating systems, use the NfcManager.requestTechnology(NfcTech.Iso15693IOS) technology and for the password:
NfcManager.iso15693HandlerIOS.customCommand(
flags = number,
customCommandCode = number,
customRequestParameters= [] );

I am using the structure

I had a problem sending the 0xB3 custom command using the handler.customCommand function.

Here is my code, I can't see any error there, but my RN code only will give me that [error] message, without giving me any information about what is going on.

try {     const idBytes= [1,2,3,4,5,6,7,8]; const passwordBytes= [0,0,0,0,0,0,0,0]
          resp = await NfcManager.iso15693HandlerIOS.customCommand({
          flags = 0x02,
          customCommandCode = 0xb3,
          customRequestParameters = [
          0x02, 0xb3, 0x02, ...idBytes, 0x02, ...passwordBytes, ],        
        });
Giving me this error:
Error [error]
Could you help me out?

Thanks!

ramazanerdogan, 2023-08-04:
```
try {     const idBytes= [1,2,3,4,5,6,7,8]; const passwordBytes= [0,0,0,0,0,0,0,0]
          resp = await NfcManager.iso15693HandlerIOS.customCommand({
          flags = 0x02,
          customCommandCode = 0xb3,
          customRequestParameters = [
          0x02, ...idBytes, 0x02, ...passwordBytes, ],        
        });
```
when i try this way i still get an error please help me @whitedogg13

jahir8a, 2023-10-11:
Hi everyone,

I hope this helps somebody, in order for you to identify the error that nfc-manager is returning as an exception, you must use an "if" and "instanceof" to know what error has occurred based on the following exceptions:

- UnsupportedFeature 
- SecurityViolation 
- InvalidParameter 
- InvalidParameterLength
- ParameterOutOfBound
- RadioDisabled
// transceive errors
- TagConnectionLost
- RetryExceeded
- TagResponseError
- SessionInvalidated
- TagNotConnected
- PacketTooLong
// reader session errors
- UserCancel
- Timeout
- Unexpected
- SystemBusy
- FirstNdefInvalid
// tag command configuration errors
- InvalidConfiguration
// ndef reader session error
- TagNotWritable
- TagUpdateFailure

For example 
`if (ex instanceof NfcError.UserCancel) {
     //bypass
} else if (ex instance of TagConnectionLost){
   //do something 
}`.

closed 2024-01-25 by github-actions[bot]