revtel/react-native-nfc-manager #654: ISO-14443 Type B support on iOS

zbelateche, 2023-07-26:
In the README, it's indicated that NFC-B is not supported on iOS.

Is this referring to the ISO/IEC 14443 Type B anticollision and modulation scheme? If I had an NFC device that used ISO/IEC 14443 Type B anticollision and modulation, would I be unable to communicate with this device using react-native-nfc-manager?

If so, is the lack of support due to fundamental hardware or software limitations of the iPhone itself, or is it just not implemented yet in react-native-nfc-manager? If it's the latter, I may start working on some of these features and submit a PR. :)

radsemijmitchell, 2023-09-05:
Based on this documentation, iOS should support ISO-14443 B so this is likely an error in the README.

https://developer.apple.com/documentation/corenfc/nfcpollingoption/nfcpollingiso14443?language=objc

[image]

radsemijmitchell, 2023-09-05:
@whitedogg13

closed 2023-12-21 by github-actions[bot]