revtel/react-native-nfc-manager #657: NfcManager.java Build Error with setNdefPushMessage

adeelcheeema, 2023-08-15:
NfcManager.java -> setNdefPushMessage ->  nfcAdapter.setNdefPushMessage(msgToPush, currentActivity);

**Issue Description:**

When trying to build the Android project, I encountered an issue related to **NfcManager.java** and the **setNdefPushMessage** function. Specifically, the build fails if I comment out the line **nfcAdapter.setNdefPushMessage(msgToPush, currentActivity)**;. However, the build succeeds when this line is included.

Expected Behavior:
The build should complete successfully even when the line nfcAdapter.setNdefPushMessage(msgToPush, currentActivity); is commented out.

Actual Behavior:
The build fails when the line nfcAdapter.setNdefPushMessage(msgToPush, currentActivity); is commented out.

`NfcAdapter nfcAdapter = NfcAdapter.getDefaultAdapter(context);
nfcAdapter.setNdefPushMessage(msgToPush, currentActivity);
callback.invoke();`

[image]

kirandchavan, 2023-08-23:
yes, this error is coming with react native version 
`"react-native": "0.72.4",`
`"react-native-nfc-manager": "^3.14.6",`

whitedogg13, 2023-08-24:
Hi @nicolasburtey @adeelcheeema @kirandchavan you're right! I have removed the deprecated api and released `v3.14.7`, please give it a try & thanks for the information!

closed 2023-08-24 by whitedogg13