revtel/react-native-nfc-manager #666: Use Tech Ndef and NfcA together on Android

LucasPiacitelli, 2023-09-23:
Hi, I hope you are doing well and thanks for the library.

About the issue, it seems that I'm unable to use Ndef and NfcA techs together. Basically, I'm getting the same errors described by @moony-chen in #661. Even if a solution was already merged in 3.14.8 release, I'm still facing those errors (`Error: unsupported tag api` and `Error: java.lang.ClassCastException: android.nfc.tech.NfcA cannot be cast to android.nfc.tech.Ndef`).

The main difference between his issue is that I'm using NXP Ntag213 to perform the tests instead of NTag215 but since I'm new to NFC technology, I can't find the exact issue.

Can anyone help me with that, please?

moony-chen, 2023-09-24:
That was just a workaround, so you still need to insert below code before you can call any api on `nfcAhandler`:
```
if (isAndroid) {
  await (NfcManager as any).close();
  await (NfcManager as any).connect([NfcTech.NfcA]);
}
```

LucasPiacitelli, 2023-09-24:
> That was just a workaround, so you still need to insert below code before you can call any api on `nfcAhandler`:
> 
> ```
> if (isAndroid) {
>   await (NfcManager as any).close();
>   await (NfcManager as any).connect([NfcTech.NfcA]);
> }
> ```

Oooh, I get it now. When I saw this snippet, I misunderstood where exactly I should insert it. I tested it and it's working fine now. Thank you so much for your help. Will come back to this issue if I need help again but, for now, I'll close it.

closed 2023-09-24 by LucasPiacitelli

ELichto, 2024-05-29:
> but with this code we have. to touch nfc chip two times to mobile, what if we want to do this in single touch.

any idea how to do this ?

ELichto, 2024-05-29:
can the operation be inverted? checking if the password is correct using NfcA then writing using Ndef? 
I tried by I got [Error: java.io.IOException] when executing writeNdefMessage, the authentification worked

`        await NfcManager.ndefHandler.writeNdefMessage(bytes);
`