revtel/react-native-nfc-manager #672: cannot convert undefined value to object.

aasharck, 2023-10-24:
Whenever I try to read an NFC tag/card. It shows me this error ```cannot convert undefined value to object.```

Any workaround for this?

crojasv90, 2023-10-26:
Maybe you are using expo Go and you do not have the plugin configured in the app.json. [Expo GO](https://github.com/revtel/react-native-nfc-manager/wiki/Expo-Go)
In my case, only NFC works for me when I build the APK, but I use expo GO, with CLI it is not necessary to build.

aasharck, 2023-10-26:
Nop. I am using react native cli. Also, I am trying to read from another device using HCE. One phone is Android 13 and the other is Android 10. I am able to read from Android 13 by using the Android 10 device, but not the other way around. 

Not sure why

crojasv90, 2023-10-26:
It seems to me that it is an NFC permissions issue. Check that part for the android version.

aasharck, 2023-10-26:
Thanks, I checked the permissions, But I don't see any problem there. NFC is allowed for the app on both phones

johncantrell97, 2023-10-26:
Are you getting this when trying to access the array `ndefMessage` on the returned tag? I have found that even though the `TagEvent` interface says it will always return an ndefMessage: NdefMessage[] it sometimes (~10-20%) does not actually contain the ndefMessage key.

For me this seems to happen randomly, not in any predictable sort of way.  I assume it's some kind of read error?

Dug a little into the source and it seems like at least on ios, if there's an error reading the ndefMessage it doesn't propagate the error but just returns the tag without the message (like I'm seeing):

```
if (!error) {
   [rnTag setObject:[NfcManager convertNdefMessage:ndefMessage] forKey:@"ndefMessage"];
}

callback(@[[NSNull null], rnTag]);
```

aasharck, 2023-10-26:
Yea exactly.

When I tried console logging "tag" variable, In the types attribute, ndef type was no found. Not sure why?

johncantrell97, 2023-10-26:
I believe it's due to an error reading the tag in native code that doesn't bubble up to the javascript.

kamatil-dev, 2024-01-14:
#692 @crojasv90 @johncantrell97 can you please help me?

closed 2024-04-30 by github-actions[bot]