revtel/react-native-nfc-manager #673: Cannot implement nfc manager in expo

christianevan, 2023-10-25:
`async function readNdef() {
    NfcManager.start();
    try {
      // register for the NFC tag with NDEF in it
      await NfcManager.requestTechnology(NfcTech.Ndef);
      console.log('Testing');
      // the resolved tag object will contain `ndefMessage` property
      const tag = await NfcManager.getTag();

      if (tag) {
      console.warn('Tag found', tag);
      } else {
        console.warn('No NDEF tag found');
      }
    } catch (ex) {
      console.warn('Oops!', ex);
      throw ex;
    } finally {
      ()=>NfcManager.cancelTechnologyRequest();
    }
  }`

I just followed the implemented code of react native nfc manager. But it always throws an error: "TypeError: Cannot convert null value to object". I use expo in my project. As you can see in my code, the console log is never shown.

closed 2023-10-25 by christianevan

nfuenmamelt, 2023-10-30:
@christianevan How do you solve this issue, I'm having the same using expo I'm trying to read or write using "await NfcManager.requestTechnology(NfcTech.Ndef);" but never end this method.

christianevan, 2023-10-30:
I haven't resolved it yet. But I guess the easiest way is to make a custom native module for the NFC reader and then use it in Expo.

You can check this: https://docs.expo.dev/modules/native-module-tutorial

christianevan, 2023-10-30:
> @christianevan How do you solve this issue, I'm having the same using expo I'm trying to read or write using "await NfcManager.requestTechnology(NfcTech.Ndef);" but never end this method.

I haven't resolved it yet. But I guess the easiest way is to make a custom native module for the NFC reader and then use it in Expo.

You can check this: https://docs.expo.dev/modules/native-module-tutorial

kamatil-dev, 2024-01-04:
any update? @christianevan @nfuenmamelt

renesass, 2024-08-25:
I don't know if it helps you but adding this to the entitlements helped me to get it running:

```
ios: {
    entitlements: {
      'com.apple.developer.nfc.readersession.formats': ['NDEF', 'TAG'],
    },
    infoPlist: {
      'com.apple.developer.nfc.readersession.formats': ['NDEF', 'TAG'],
    },
  },
  ```
  
  Also, remove your ios folder to make sure the entitlements file is updated. Hope it helps!