revtel/react-native-nfc-manager #674: Alert when disconnecting NFC card

crojasv90, 2023-10-26:
Greetings.
I need to show one alert when an NFC card is connected and another when it is disconnected.
I can't achieve the disconnection, I tried with setEventListener with the sessionClosed event, but it never shows the disconnection.
Does anyone know how to obtain the event?

alan-dev-hk, 2023-11-07:
Hey, man
Could you find the solution?

crojasv90, 2023-11-07:
I'm still looking for a solution. Do you have any idea? Or do you know if the technology supports it?

alan-dev-hk, 2023-11-07:
No, unfortunately. I'm still look for some solution as well. In case I resolve it, I post the here.

alan-dev-hk, 2023-11-07:
Hey, maybe I finded the solution. `NfcManager.setEventListener(NfcEvents.StateChanged, data => this.CallbackDisconect(data));` So, the NfcEvents.StateChanged will fire when you turn on or turn of the NFC. So, maybe you can work with it. 

The responses are: 'on' | 'turning_on' | 'off' | 'turning_off'

closed 2024-02-21 by github-actions[bot]