revtel/react-native-nfc-manager #677: ISO 14443-4 not reading in IOS

Tboules, 2023-11-02:
Hey y'all,

just got a batch of NFC chips from a manufacturer that are ISO 14443-4 and they are not registering in IOS. I am able to read them on Android with the app I built but not IOS. 

Here is an image of the tag details

[image]

type ISO 14443 - 3A seem to work fine

closed 2024-02-15 by github-actions[bot]

Danushka96, 2025-01-30:
Same here, tried exact card type from android and it's working, but in ios its not detecting

miklund, 2025-11-25:
Same issue here. Did you find out a solution or a workaround?

```
await NfcManager.requestTechnology(NfcTech.Ndef);
const tag = await NfcManager.getTag();
console.log('Tag metadata:', tag);
const data = await NfcManager.ndefHandler.getNdefMessage();
```

`getTag()` returns an empty object, and `getNefMessage()` throws an error with message "No ndef available".

I've also tried requestTechnology `IsoDep` and `transceive` but this fails with `Not connected`.