revtel/react-native-nfc-manager #686: Very poor documentation

LPitonakova, 2023-12-12:
Thanks for sharing the library with us, but the documentation could use some serious improvements. There is no API documentation and no way for us to know what the options for creating different records are. Even the examples of reading / writing are not complete.

I am having to look directly in your code to see the "API", but there is no documentation of your functions. So it's hard to figure out sometimes what the function parameters should be.

If you could at least document all your functions that would be most helpful.

cavasinf, 2024-01-18:
I strongly recommend to go and grab the NfcProxy class from the demo application.
https://github.com/revtel/react-native-nfc-rewriter/blob/master/src/NfcProxy.js

You can use it directly or take some code from there.

Aiosa, 2024-02-24:
Although this is somewhat helpful, still, how does one react on card detected event? Does not seem to support. The proxy would be good if it had all the features, but what if I want to do something on card detection? Android will by default show the loaded tag. :/ I want the app to react instead.

cavasinf, 2024-02-24:
@Aiosa this is not related to this library, but you need to use deep links in you app, and tell it can certain links. 
Then write that link in your nfc tag.
When the phone will try to open your link, it will ask to every app on the phone if someone as deep link to it.
Yours should respond with an intent filter.

An example can be Google map, when clicking a link, it will prompt you to open it with a navigator, or directly with maps.

Aiosa, 2024-02-24:
Thanks for reply. I meant the library ability to detect the tag automatically in the first place without user having to tap any button (while having the app opened). I found the event usage in the proxy afterwards, but I had to really dig deep :/ It has not comments so I still agree some light docs would rock

closed 2024-06-09 by github-actions[bot]