revtel/react-native-nfc-manager #688: tag.ndefMessage is undefined

mickytroxxy, 2023-12-20:
Hello my fellow devs, I have been stuck for almost 48 hours now on the tag.ndefMessage which is being undefined, but the tag itself has  Object {
  "id": "0E9A3923200B20",
  "techTypes": Array [
    "android.nfc.tech.IsoDep",
    "android.nfc.tech.NfcA",
  ],
}

Below is my code, I tried using rn-cli then moved to eas-build with no luck

import React, { useEffect, useState } from 'react'
import { Text, View, TouchableOpacity } from 'react-native';
import { LinearGradient } from 'expo-linear-gradient';

import * as Animatable from 'react-native-animatable';
import Icon from '../components/ui/Icon';
import { colors } from '../constants/Colors';
import { useRouter } from 'expo-router';
import { GlobalStyles } from '../styles';
import NfcManager, {NfcTech} from 'react-native-nfc-manager';
NfcManager.start();
const SplashScreen = () => {
  const router = useRouter();
  const [nfcData, setNfcData] = useState(null);
  useEffect(() => {
    NfcManager.start();
    return () => {
      NfcManager.cancelTechnologyRequest();
    };
  }, []);

  const readNdef = async () => {
    try {
      console.log('Waiting for the read');
      await NfcManager.requestTechnology([
        NfcTech.MifareIOS,
        NfcTech.Ndef,
        NfcTech.Iso15693IOS,
        NfcTech.IsoDep,
        NfcTech.NfcA,
        NfcTech.NfcB,
        NfcTech.NfcF,
        NfcTech.NfcV,
        NfcTech.FelicaIOS,
        NfcTech.MifareClassic,
      ]);
  
      const tag = await NfcManager.getTag();
      console.warn('Tag found', tag);
      if (tag.ndefMessage) {
        tag.ndefMessage.forEach((record) => {
          console.log('NDEF Record:', record);
          const payload = Ndef.text.decodePayload(record.payload);
          console.log('Record Payload:', payload);
        });
      } else {
        console.warn('Tag does not contain NDEF message');
      }
    } catch (ex) {
      console.warn('Oops!', ex);
    } finally {
      NfcManager.cancelTechnologyRequest();
    }
  };
  return(
    <View style={GlobalStyles.container}>
        <View style={GlobalStyles.header}>
            <Animatable.Image animation="bounceIn" duration={1500} useNativeDriver={true} source={require('../assets/images/heartios.png')} style={GlobalStyles.logo} resizeMode="stretch"/>
        </View>
        <Animatable.View animation="fadeInUpBig" useNativeDriver={true} style={GlobalStyles.footer}>
            <Text style={[GlobalStyles.title,{fontFamily:'fontBold'}]}>All For Your Beauty</Text>
            <Text style={[GlobalStyles.text,{fontFamily:'fontBold'}]}>Request for hair stylist, coiffurist and buy exotic cosmetic products</Text>
            <View style={GlobalStyles.button}>
                <TouchableOpacity onPress={()=>readNdef()}>
                    <LinearGradient colors={["#e44528","#d6a8e7","#f3bf4f"]}start={{ x: 0, y: 1 }} end={{ x: 1, y: 0 }} style={GlobalStyles.signIn}>
                        <Text style={[GlobalStyles.textSign,{fontFamily:'fontBold'}]}>Get Started</Text>
                        <Icon name='navigate-next' type='MaterialIcons' size={20} color={colors.white} />
                    </LinearGradient>
                </TouchableOpacity> 
            </View>
        </Animatable.View>
    </View>
  )
}
export default SplashScreen

Please help

siash1, 2023-12-22:
same issue, please help

siash1, 2023-12-22:
Same problem please help

closed 2024-04-07 by github-actions[bot]