revtel/react-native-nfc-manager #692: Unable to write

kamatil-dev, 2024-01-14:
Hello guys, please i've contacted so many devs but none of them were able to solve this problem

you can check this snack [https://snack.expo.dev/@karim.amahtil/bossy-red-chocolates](https://snack.expo.dev/@karim.amahtil/bossy-red-chocolates) and tell me why the writing is not working (you can go directly to line 80), i've tried expo cloud build and local build, on both the writing function was not working

kamatil-dev, 2024-01-14:
@whitedogg13

cavasinf, 2024-01-18:
Did you take a look at https://github.com/revtel/react-native-nfc-manager/wiki/Examples#write-a-tag?

```js
if (bytes) {
  await NfcManager.ndefHandler // STEP 2
    .writeNdefMessage(bytes); // STEP 3
  result = true;
}
```

Your `if` is inverted:
> if (!bytes) await NfcManager.ndefHandler.writeNdefMessage(bytes);

closed 2024-01-23 by kamatil-dev