revtel/react-native-nfc-manager #695: Update dependency version for @expo/config-plugins

lawnguyen, 2024-01-22:
Need to update to support @expo/config-plugins:"~7.8.0".

RonakDoshiTMI, 2024-01-26:
+1

[image]

hezzze, 2024-02-27:
+1

[image]

nfuenmamelt, 2024-03-12:
any solutions for this

mwiedemeyer, 2024-06-10:
Can someone please update the @expo/config-plugins dependency to 8.0.0? Its required for Expo SDK51
[image]

See: https://github.com/revtel/react-native-nfc-manager/blob/9301ab3f8d13115614d11ea6f1f9da22697835f1/package.json#L65

fredlemieux, 2024-07-09:
Please see PR #732, tests pass, please let me know if there is anything else I need to do?  Do I need to bump the version number? I looks like this is done automatically using "release-it"

fredlemieux, 2024-07-09:
There are loads of outdated packages and vulnerabilities, I am happy to raise another PR fixing these once this is merged, which is a priority...

fredlemieux, 2024-07-09:
> There are loads of outdated packages and vulnerabilities, I am happy to raise another PR fixing these once this is merged, which is a priority...

#733 <- Here it is...

closed 2024-10-24 by github-actions[bot]