revtel/react-native-nfc-manager #704: Enabling Multiple Uses of NFC requestTechnology Instead of One-time

divashuthron, 2024-03-12:
Hello. I am an IT developer working for a small company in Korea. 

What I want to achieve is a functionality where, regardless of whether NFC is tapped once or not, pressing a 'Refresh' button turns off NFC reading and then turns it back on when pressed again. 

However, I encountered an error after implementing this. 

Could you please let me know if there is anything unusual in my source code?

`const nfcTagLogin = async () => {
    try {
      NfcManager.start();
      await NfcManager.requestTechnology(NfcTech.Ndef);
      const tag = await NfcManager.getTag();
      // console.log('NFC tagged, ID: ' + tag.id);

      webviewRef.current.postMessage(String(tag.id));
    } catch (error) {
      console.error(error);
    } finally {
      NfcManager.cancelTechnologyRequest().catch(() => 0);
    }
  };

const nfcClean = () => {
    // 실행중인 NFC 이벤트, Request가 있는 경우 종료
    NfcManager.unregisterTagEvent();
    NfcManager.cancelTechnologyRequest();
    NfcManager.stop();
  };
`

divashuthron, 2024-03-12:
const nfcTagLogin = async () => {
    try {
      NfcManager.start();
      await NfcManager.requestTechnology(NfcTech.Ndef);
      const tag = await NfcManager.getTag();
      // console.log('NFC tagged, ID: ' + tag.id);

      webviewRef.current.postMessage(String(tag.id));
    } catch (error) {
      console.error(error);
    } finally {
      NfcManager.cancelTechnologyRequest().catch(() => 0);
    }
  };

  const nfcClean = () => {
    NfcManager.unregisterTagEvent();
    NfcManager.cancelTechnologyRequest();
  };

  const nfcRefresh =() => {
    nfcClean();
    nfcTagLogin();
  };

georgipavlov-7DIGIT, 2024-04-25:
I think the problem comes from trying to start the scanning immediatly after the cancellation. I had a similar problem and solved it by adding a small delay, which in your case will look like this

`
const nfcRefresh = () => {
  NfcManager.cancelTechnologyRequest().then(async () => {
    await new Promise((resolve) => setTimeout(resolve, 3500)).then(() => {
      nfcTagLogin()
    });
  })
}
`

There might be a better solution, but this it the one that i managed to come up with, hope it helps you

closed 2024-08-09 by github-actions[bot]