revtel/react-native-nfc-manager #709: Ndef not writing the empty tag in android only

Nardeep, 2024-03-27:
When tag is empty , then  ndef.writeNdefMessage is not working its throwing error :
[Error: unsupported tag api]

we are using below code:
const writeDataToNfcTag = async (data) => {
        console.log('data-=-=-',data);
        try {
          await NfcManager.requestTechnology(NfcTech.Ndef, {
                alertMessage: 'Ready to write some NDEF',
            });
            const bytes = Ndef.encodeMessage([Ndef.textRecord(data)]);
            console.log('bytes-=-=',bytes);
            if (bytes) {
             try{
                await NfcManager.ndefHandler.writeNdefMessage(bytes);

                if (Platform.OS === 'ios') {
                    await NfcManager?.setAlertMessageIOS('Successfully write NDEF');
                }
             }catch(err){
                console.log('writeError-=-',err);
             }

            }

        } catch (error) {
            console.error('Error writing data to NFC tag:', JSON.stringify(error));
        } finally {
            await NfcManager.cancelTechnologyRequest();
        }
    }
    
    
    After debugging the android code we found that when tag is empty then its not detecting  techRequest.
    Ndef ndef = (Ndef)techRequest.getTechHandle();
    ndef is coming null. So please check why its happening

K-Leon, 2024-04-07:
Same issue on our side. Did you find a solution / workaround?

Nardeep, 2024-07-08:
@K-Leon Hi , we resolved this issue. The issue is in technology type we added the technology and use the write function according to nfc technology. Here is the below code:
  let technology = await NfcManager.requestTechnology([NfcTech.NdefFormatable, NfcTech.Ndef], 

let promise = (technology == "NdefFormatable" ? NfcManager.ndefFormatableHandlerAndroid.formatNdef(bytes) : NfcManager.ndefHandler.writeNdefMessage(bytes))

closed 2024-10-23 by github-actions[bot]