revtel/react-native-nfc-manager #710: ISO15693 customCommands & password protection not working

outboundexplorer, 2024-04-07:
I have been trying to set password protection for the ICODE SLIX.

The data set here at [dataset](https://www.nxp.com/docs/en/data-sheet/SL2S2002_SL2S2102.pdf).

I am using the NfcManager.iso15693HandlerIOS.customCommand() as I am currently developing on IOS.

The react-native-nfc-manager has been fantastic to work with type 2 tags.  But really running into problems on setting password protection for the type 5 tags.

As i understand it, I need to follow the following steps to add password protection to ISO15693 tags.

1. Generate random number
2. Set password
3. Write password
4. Lock password
5. Enable password protection

Here is an example of a command that I am using just to generate a random number.

```javascript
try {
    const tag = await NfcManager.getTag();
    const icMfgCode = 0x04; // Assuming NXP's manufacturer code
    const uidBytes = tag?.id.match(/.{1,2}/g).map(byte => parseInt(byte, 16));
    const response = await NfcManager.iso15693HandlerIOS.customCommand({
      flags: 0x02,
      commandCode: 0xb2,
      data: [icMfgCode, ...uidBytes],
    });

    console.log('response for random number:', response);
  } catch (error) {
    console.error('Failed to get random password:', error);
  } finally {
    NfcManager.cancelTechnologyRequest().catch(err => console.error(err));
  }
```

[image]

This is the structure for the command as defined by the product data sheet.

psusundrem, 2024-05-30:
[image]

Avoid using the 64-bit optional UID in the command with the 0x02 flag. If you still want to use a UID, try using the flags: 0x22 and data: [icMfgCode, ...uidBytes.reverse()]. The UID bytes are fetched in reverse order.

Here’s how I use a custom command for the "present password" command for an ST type 5 chip:
const checkPasswordIosCmd = {
  flags: Nfc15693RequestFlagIOS.HighDataRate, //Bit 2 SET, High data rate
  customCommandCode: 0xb3, //Present Password Command code
  customRequestParameters: [0x01, 0x55, 0xAA, 0x55, 0xAA, 0xD7, 0xD3, 0x03, 0xE7],
};
const resp = await NfcManager.iso15693HandlerIOS.customCommand(checkPasswordIosCmd);

In this command, the customRequestParameters array starts with the RF password number, followed by the 8-byte actual password.

outboundexplorer, 2024-06-02:
I will give this a try.  Thanks.

closed 2024-09-17 by github-actions[bot]